ami-iit/matlab-whole-body-simulator #4: Duplicated functions/tools in the repo

nunoguedelha, 2020-10-13:
Hi @Giulero , I noticed the minor following problems with the tools in `src/tools`:
- The tools `expinential_map`, `H2Rp`, Rp2H` are redefined in the classes. I guess their implementation in the folder `src/tools` was just the first step...
- In the classes, you sometimes use `wbc.skew`, and sometimes use `skew`. I guess the second case is a leftover right?

As a consequence, I suggest we remove all of them except `get_R_dot`, which actually isn't used anywhere yet, unless I missed something.

I can do it in my PR if you agree.

nunoguedelha, 2020-10-13:
More precisely, the 3 mentioned tools are defined as static methods of class `State`.

Giulero, 2020-10-13:
Hi @nunoguedelha! Sure! You could remove also `get_R_dot`. I was using it when $\dot{R}$ was integrated, while now the simulator uses the exponential map.

nunoguedelha, 2020-10-13:
Great!

nunoguedelha, 2020-10-14:
Fixed by #3 .

commit f8f4e0fb (2020-10-13): Fixed #4.

- Cleaned-up duplicated and unused tools in folder `src/tools`.
- Replaced `skew` occurrencies by `wbc.skew`.
PR #3 (2020-10-13): Make a library out of 'robot' simulator with contacts sub-system
The goal is to make a library (in the Simulink sense) out of the sub-system simulating the robot dynamics with feet/ground contacts handling.

## The changes are listed below:
- Minor refactoring: Move the joint torque delay block and the Robot Visualizer outside the sub-system.
- Create `robotDynamicsWithContacts_lib.slx` as a copy of the refactored `robot` sub-system.
- Implement required changes allowing to display the library `robotDynamicsWithContacts_lib.slx` in the Simulink library browser (e.g. `slblocks.m`).
- Moved addi [... cut]
commit b619a313 (2020-10-13): Fixed #4.

- Cleaned-up duplicated and unused tools in folder `src/tools`.
- Replaced `skew` occurrencies by `wbc.skew`.
closed 2020-10-14 by nunoguedelha