ami-iit/matlab-whole-body-simulator #5: Improve whole-body and contacts simulator core execution speed

nunoguedelha, 2020-10-21:


nunoguedelha, 2021-01-13:
## Proposed solution
- Replace bindings by Simulink functions;
- These functions trigger Optimized blocks “code generation ready”;
- Requires creation of a few WBT/Factory blocks as iDynTree proxies;
- Factorize calls to these WBT blocks.

We cannot replace the bindings by Simulink function calls combined with Simulink WBT blocks in a one-to-one mapping basis, mostly for the following reasons:
- it would require a great number of additional WBT blocks to be created,
- the interface of such blocks only supports numerical inputs/outputs, so it does not support objects, while many functions in the usual bindings return or take objects as inputs (Swig objects, etc).
- after a quick analysis of the `RobotDynamicsWithContacts` code, we can realize that the simplest approach for now is to address the substitution of the bindings in a case by case basis.

## `RobotDynamicsWithContacts` design aspects to consider
From the `RobotDynamicsWithContacts` code analysis we identified the following aspects to take in account in the substitution.
- The only class among the three (`Robot`, `Contacts`, `State`) to use bindings is the class `Robot` (except eventually for some isolated case)
- Most of the used bindings are methods of the class `iDynTree::KinDynComputations`, called through an object created in `iDynTreeWrappers.loadReducedModel()` function which loads the robot model from a file.
- The loading of the robot model is actually already done by any existing WBT block implementation through the call of `RobotInterface::getKinDynComputations()`, which either initializes a new `KinDynComputations` object, either reuses an existing one, which is equivalent to the usage of a singleton.

So we just need to use a `Configuration` WBT block in order to handle the robot model loading, and create new blocks, if needed, for replacing the used bindings within `KinDynComputations`.

nunoguedelha, 2021-01-13:
## Implementation steps

So the following steps are:
- [ ] Add a `Configuration` block in the main Simulink test model and set its configuration as usual, through a variable of class `WBToolbox.Configuration`.
- [ ] Make it clear in the `RobotDynamicsWithContacts` library interface that it depends on a `Configuration` block. This can easily be done through a mask which would include, in addition to the other input parameters, a toggle button "Toggle Config block highlighting".
- [ ] List all the called iDynTree bindings methods, either directly called through the `iDynTree` object, either called through the `iDynTree::KinDynComputations` object.
- [ ] Replace the binding methods already implemented in an existing WBT block.
- [ ] Create WBT blocks for the missing methods and replace the respective bindings.

nunoguedelha, 2021-01-13:
CC @CarlottaSartore @Giulero

nunoguedelha, 2021-01-13:
## WIP: List all the called iDynTree bindings methods

List all the called iDynTree bindings methods, either directly called through the iDynTree object, either called through the `iDynTree::KinDynComputations` object.

| method | Existing | To Be implemented |
| --- | --- | --- |
| iDynTree.FreeFloatingGeneralizedTorques | InverseDynamics | |
| iDynTree.ModelLoader | * | |
| iDynTree.ModelLoader.model | * | |
| iDynTree.KinDynComputations | * | |
| kinDynComp.getFrameIndex | ** | |
| kinDynComp.getFreeFloatingMassMatrix | MassMatrix | |
| kinDynComp.generalizedBiasForces | GetBiasForces | |
| kinDynComp.getFrameFreeFloatingJacobian | Jacobian | |
| kinDynComp.getFrameBiasAcc | | x |

(*) WBT shared KinDynComputations
(**) Can be avoided by using durectly the frame name (string) as a block parameter if this parameter is set once at initialization

nunoguedelha, 2021-01-14:
- [ ] iDynTree.ModelLoader
- [ ] iDynTree.ModelLoader.model
- [ ] iDynTree.KinDynComputations.setRobotState
- [ ] iDynTree.KinDynComputations.getFrameIndex
- [ ] iDynTree.FreeFloatingGeneralizedTorques
- [ ] iDynTree.KinDynComputations.getFreeFloatingMassMatrix
- [ ] iDynTree.KinDynComputations.generalizedBiasForces
- [ ] iDynTree.KinDynComputations.getFrameFreeFloatingJacobian
- [ ] iDynTree.KinDynComputations.getFrameBiasAcc
- [ ] iDynTree.KinDynComputations.getWorldTransform
- [ ] iDynTree.KinDynComputations.setFloatingBase

nunoguedelha, 2021-01-28:
Now Fixing compilation & run-time errors:
- OSQP Matlab bindings not supported, nor `quadprog`. Use qpOASES Simulink block instead through a Simulink function `simFunc_qpOASES` sub-system. Selection is done through `setOSQP` and `useQPOASES` bool flags.
- Some fields of the `robot_config` input of `step_block` Matlab System are modified or modifiable in run-time, so they
  cannot be Tunable. Don't set this input to `Nontunable`.
- `Nontunable` inputs have to be initialized.
- Cleanup `x_iDyn` variables formerly used for interfacing bindings functions.
- Matrices need to be properly initialized (sparse iterative filling of matrix, as done in Matlab, is not supported,
  dynamic size matrices).
- Unsupported features or functions like `optimoptions`, `quadprog` or `osqp` binding will cause the code generation
  compiler to fail even if they are wrapped in a if or switch condition.
- Other features not supported in code generation are: `mat2cell`, implicit indexing ":".

nunoguedelha, 2021-02-01:
Fixed other size mismatch errors by:
- replacing implicit indexing (`:`, `1:end`) by explicit indexing.
- replacing variables used in explicit indexing by constants when necessary.
- using constants in "if" conditions instead of variables, in order to make sure that we do not attempt to compile conditional code which is not supported by the code generator.
- refactor patterns not supported by the code generator (e.g. `A = []; A = [A;B]`).
- Remove persistent variables.

The Simulator now runs, but the robot is not moving in the visualizer.

nunoguedelha, 2021-02-01:
## Issue Analysis: Robot not moving through the whole simulation

For debugging the issue, we set the initial robot base pose to a significant height (e.g. 10m), in order to check a simple scenario, without link/ground impacts:
```
initialConditions.base_position = [0; 0; 10.65];
```
As the Matlab System block is now compiled with code generation, the usual MATLAB breakpoint workflow is not available, so we use the MATLAB display command to log critical computed variables. This way, we compare the dynamic quantities computed in `step_block.stepImpl` between the current version and the `devel` branch.

The target quantities are the outputs of `obj.contacts.compute_contact()`  and `obj.robot.forward_dynamics()`. As expected, we observe significant differences on the `obj.robot.forward_dynamics()` outputs: `base_pose_ddot`, `s_ddot`, and more precisely, the computation of the bias forces `h = obj.get_bias_forces()`. The faulty code is computing `h` as all zeros instead of the proper values.

nunoguedelha, 2021-02-01:
## Issue Analysis: Robot not moving through the whole simulation

> There is probably an issue with the WBT `GetBiasForces` block configuration.
> 
> Looking now closer at the problem...

Just found the bug. Because of a double definition, the passed gravity vector was `[0 0 0 0 -9.81]` instead of `[0 0 -9.81]`, so the resulting gravity value was null as only the first 3 passed components are considered valid.

nunoguedelha, 2021-02-01:
Implemented in https://github.com/dic-iit/matlab-whole-body-simulators/pull/15, with the fixes described above.

nunoguedelha, 2021-02-01:
## Issue: When hitting the round, the visualizer crashes due to singular matrix computation

[image]

The issue is actually caused by the fact that the robot right foot sinks in the ground (contact failure), creating very high reaction forces causing the simulation to diverge.

Looking now for the root cause of the foot contact failure.

nunoguedelha, 2021-02-02:
> Looking now for the root cause of the foot contact failure. Check:
> 
> - Contact points before and after impact in compute_contact_points()

- Flag `left_right_foot_in_contact` is set to 1 when the right foot gets in contact with the ground at `t=0.272 sec`.

[image]

- The mapping of the new contact points was wrong. A new contact is considered when a vertex was not in contact and now is: `mapVerticesNewContact = obj.is_in_contact & ~obj.was_in_contact`
- The computation of the respective indexes in `J_feet` was also wrong.

Fixed in https://github.com/dic-iit/matlab-whole-body-simulators/pull/15/commits/228d700a3687a3463aea452f958e5daaf102538b of https://github.com/dic-iit/matlab-whole-body-simulators/pull/15.

nunoguedelha, 2021-02-02:
## Issue: Only one of the feet is staying "above the ground"

After the robot sinks in the ground, only the vertices of the right foot stay on the ground, the other foot sinks far under the ground level.

[image]

This is probably related, again, with the contact mapping/index computations and velocities corrections. We'll check:

- [x] Contact points before and after impact in `compute_contact_points()`
- [ ] Velocities correction in `compute_velocity()`
- [ ] `forces` computation in `compute_unilateral_linear_contact()`
- [ ] `A` computation in `compute_unilateral_linear_contact()`

CarlottaSartore, 2021-02-02:
C.C. @VenusPasandi

nunoguedelha, 2021-02-03:
Last issue fixed in https://github.com/dic-iit/matlab-whole-body-simulators/pull/15/commits/80e74c96df1424765c9dbbfee1bd1e28175face1 of the ongoing pull request.

nunoguedelha, 2021-02-04:
## Execution speed analysis using the Simulink profiler:

Actually, the last bottleneck slowing down the simulation was the visualizer, as we'll show in the following analysis. We illustrate below the blocks we will focus on in the execution profiler...

The `Robot visualizer/MATLAB System` and 9 Simulink functions:

[image]

The `Robot visualizer/MATLAB System` in the `Robot visualizer` subsystem:

[image]

We obtained the following profile for a total simulated time of 6s:

[image]

The step time used for the visualizer (`confVisualizer.tStep`) is 40ms, the total simulation time 6s.

As we can see in the profile report, it took ~48s for simulating 6s, so a rate of 1/8. The time shares for the blocks `RobotDynWithContacts/MATLAB_System`, `Robot visualizer/MATLAB_System`, `RobotDynWithContacts/Simulink_Function8` are:

| Block | Time share |
| --- | --- |
| test_matlab_system_19 run time | 69.2 % |
| Robot visualizer/MATLAB System | 49.6 % |
| RobotDynWithContacts/MATLAB System | 18.5 % |
| RobotDynWithContacts/Simulink Function8 (QP) | 2.0 % |

The time share for the other Simulink functions (Function, Function1, Funciton2, ... Function7) is comparable to the QP function. The individual share of the QP is negligible w.r.t. the `Robot visualizer/MATLAB System`.

traversaro, 2021-02-04:
> **Actually, the last bottleneck slowing down the simulation was the visualizer**, as we'll show in the following analysis.

Related PR: https://github.com/robotology/idyntree/pull/744 .

nunoguedelha, 2021-02-04:
Indeed, with a faster visualizer we could increase the display frequency and get a smoother motion in the visualizer.

nunoguedelha, 2021-03-15:
The MATLAB visualizer improvement will be dealt with in https://github.com/robotology/idyntree/pull/744.

The other improvements have been implemented in https://github.com/dic-iit/matlab-whole-body-simulator/pull/15. Closing this issue.

closed 2021-03-15 by nunoguedelha