ami-iit/matlab-whole-body-simulator #11: Discuss on the usage of tags for major, minor releases and patches in this repo

nunoguedelha, 2020-11-30:
@Giulero , @traversaro , @gabrielenava , @CarlottaSartore , as this repo is getting more visibility and started to be used in other projects, we should start using tags, what do you think?

Following the versioning convention Major.Minor.Patch, we could start by applying the tag `v0.1.0` to the commit 30642d6c195ae1c663d6ebbf43e81f95eff1946a (latest `master`).

Then, how should we proceed, tag Major/Minor new releases on `master` only, and Patch releases on `devel`?

traversaro, 2020-11-30:
My suggestions is to use https://semver.org/, and only do releases/tags on the `master` branch, or any other branch that you choose as "stable", never doing tags/release on devel/unstable branches. I also suggest to mantain a CHANGELOG : https://keepachangelog.com/en/1.0.0/ .

nunoguedelha, 2021-01-08:
> My suggestions is to use https://semver.org/, and only do releases/tags on the `master` branch, or any other branch that you choose as "stable", never doing tags/release on devel/unstable branches. I also suggest to mantain a CHANGELOG : https://keepachangelog.com/en/1.0.0/ .

@traversaro , would it be appropriate to use a pre-release tag on an unstable branch like `devel`, as [this type of tag](https://semver.org/#spec-item-9) is supposed to be used in unstable versions preceding a respective release?
```
A pre-release version indicates that the version is unstable and might not satisfy the intended compatibility
requirements as denoted by its associated normal version. Examples: 1.0.0-alpha, 1.0.0-alpha.1, 1.0.0-0.3.7, 
1.0.0-x.7.z.92, 1.0.0-x-y-z.–.
```

nunoguedelha, 2021-01-12:
Applied the first [release tag](https://github.com/dic-iit/matlab-whole-body-simulators/releases/tag/v1.0.0).

nunoguedelha, 2021-01-12:
As discussed offline with @traversaro , we can apply a pre-release tag on an unstable branch like `devel`. Such tag will follow the format recommended [here](https://semver.org/#spec-item-9).

I think we can close this issue.

closed 2021-01-13 by nunoguedelha