ami-iit/matlab-whole-body-simulator #18: Add example of simulating a pendulum

traversaro, 2021-03-11:


nunoguedelha, 2021-03-22:
## Add to the models available in the simulator the model of a pendulum.

nunoguedelha, 2021-04-01:
> Add to the models available in the simulator the model of a pendulum.

Implemented in #22 .

nunoguedelha, 2021-04-01:
## Remaining issues are...

- [ ] Fix issue with the meshes: convert from dae to stl.
- [ ] Fix meshes location such that specifying the meshes prefix is not required.

## Detailed steps

### Fix issue with the meshes: convert from dae to stl.

- [ ] Converting Collada (.dae) to URDF (.stl)

  [Collada URDF](https://github.com/ros/collada_urdf) contains packages for converting from collada files to URDF. Collada to urdf command-line tool:
  ```
  rosrun collada_urdf collada_to_urdf <input.dae> <output.urdf>
  ```
  
  `collada_urdf` package has to be built and installed...
  
  - [ ] Create a Catkin workspace, build catkin package and install in workspace (https://wiki.ros.org/catkin/Tutorials).
    
   > 1. Creating a workspace for catkin
   > 
   > 2. This tutorial covers how to setup a catkin workspace in which one or more catkin packages can be built.
   >    Creating a ROS Package
   > 
   > 3. How to create a new ROS package using catkin.
   >    Building and using catkin packages in a workspace
   > 
   > 4. How to use catkin_make to build and install packages from source in a catkin workspace.
   >    Overlaying with catkin workspaces
   > 
   > 5. Describes how to overlay a catkin workspace on to an existing system for development and testing. 

    - [ ] Install `catkin` tools, from ROS packages (e.g. [robostack](https://github.com/RoboStack/ros-noetic)).
      Install [robostack](https://github.com/RoboStack/ros-noetic):
      ```
      conda create -n robostackenv python=3.8
      conda activate robostackenv
      conda config --env --add channels conda-forge
      conda config --env --add channels robostack
      conda config --env --set channel_priority strict
      mamba install ros-noetic-desktop
      mamba install compilers cmake pkg-config make ninja catkin_tools
      conda deactivate
      conda activate robostackenv
      ```

### Fix meshes location such that specifying the meshes prefix is not required.

...

nunoguedelha, 2021-04-08:
All completed. Closing.

closed 2021-04-08 by nunoguedelha

traversaro, 2021-04-08:
Could it make sense to add some documentation in the README or linked from the readme on how to try this example? Otherwise it could be a bit difficult for users to find it.

nunoguedelha, 2021-04-14:
> Could it make sense to add some documentation in the README or linked from the readme on how to try this example? Otherwise it could be a bit difficult for users to find it.

Fixed in #33 .