ami-iit/matlab-whole-body-simulator #19: Run the simulator with the robotology binaries installed through Conda

nunoguedelha, 2021-03-26:
Since https://github.com/robotology/robotology-superbuild/pull/652, we should be able to install the `matlab-whole-body-simulator` dependencies from the Conda robotology channel, using the **conda-forge** or **Mamba** installer.

This would significantly simplify the installation of the `matlab-whole-body-simulator` simulator on  local host and unlock its easy setup and run from MATLAB online.

In view of validating such procedure, we shall:
- Test the installation on a local machine using the dependency binaries from conda robotology channel.
- Test the installation on MATLAB online as explained in https://github.com/robotology/robotology-superbuild/pull/652#issuecomment-794027294.
- Fix any encountered issues.
- Document the installation README accordingly.

nunoguedelha, 2021-03-26:
# WIP:

## Test the installation on a local machine using the dependency binaries from conda robotology channel

We followed the steps listed below:
1. Install Mamba, create a new environment and install the robotology dependency binaries:
    ```
    $ conda install mamba
    $ conda create -n robotologyenv
    $ conda activate robotologyenv
    $ mamba install -c robotology iDynTree qpOASES icub-models wb-toolbox whole-body-controllers
    ```
5. Checks... the MATLABPATH env should now have:
    ```
    <user-home-dir>/miniforge3/envs/robotologyenv/mex: <user-home-dir>/miniforge3/envs/robotologyenv/share/WBToolbox: <user-home-dir>/miniforge3/envs/robotologyenv/share/WBToolbox/images
    ``` 
    Mex libraries:
    ```
    $ ls <user-home-dir>/miniforge3/envs/robotologyenv/mex/
    ```
    +iDynTree
    +iDynTreeWrappers
    +wbc
    BlockFactory.mexmaci64
    BlockFactory.tlc
    SwigGet.m
    SwigMem.m
    SwigRef.m
    iDynTreeMEX.mexmaci64
    mesh2tri.m
    ```
6. Clone the repository  `matlab-whole-body-simulators`
    ```
    $ git clone https://github.com/dic-iit/matlab-whole-body-simulator.git
    ```
7. Run matlab in the same conda environment.
8. Change working directory to the root path of repository `matlab-whole-body-simulators`
9. Open and run the model `test_matlab_system_2020b.mdl`.

### Notes
After the environment is activated (`conda activate robotologyenv`), the terminal prompt starts with `(robotologyenv) `.

## Issues
### README
The two dependencies  **wb-toolbox** and **whole-body-controllers** were missing from the dependencies list in the repo main README.

> - [Matlab/Simulink 2019b](https://it.mathworks.com/products/matlab.html)
> - [iDynTree](https://github.com/robotology/idyntree)
> - [OSQP](https://github.com/oxfordcontrol/osqp.git)
> - [osqp-matlab](https://github.com/oxfordcontrol/osqp-matlab) (the OSQP MATLAB bindings).
> - [qpOASES](https://github.com/robotology-dependencies/qpOASES)
> - [icub-models](https://github.com/robotology/icub-models) to access iCub models.

**OSQP** and **osqp-matlab** are not needed anymore, at least until OSQP replaces qpOASES.

The main README shall be updated accordingly.

## MATLAB
[image]
...

nunoguedelha, 2021-03-26:
### 2. MATLAB issue
[image]

* => In the `getMeshes(...)` function, we were skipping the loop where the output `map` was set.
    * => This was due to the number of links of the model being 0, which was due to a wrong model path:
      ```
      robot_config.modelPath = '/share/iCub/robots/iCubGenova04/'
      ```
        * => due to `icubModelsInstallPrefix` being empty.
            * => due to environment variable `ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX` not properly set.

I was expecting this variable to be automatically set by the activation of the Conda environment.

Setting that variable would make the use of the superbuild binaries backward compatible with software depending on a locally built superbuild. What do you think @traversaro ?

nunoguedelha, 2021-03-26:
> Setting that variable would make the use of the superbuild binaries backward compatible with software depending on a locally built superbuild. What do you think @traversaro ?

I understand that it should be more robust to use more relevant variables already set in the scripts `conda/multisheller/<component>_activate.msh`, right?

traversaro, 2021-03-26:
> 
> 
> > Setting that variable would make the use of the superbuild binaries backward compatible with software depending on a locally built superbuild. What do you think @traversaro ?
> 
> I understand that it should be better to use as much as possible more relevant variables already set in the scripts `conda/multisheller/<component>_activate.msh`, right?

Yes, to be honest the `ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX` env variables was just an helper variable used inside the setup scripts, and not something that was supposed to be used. In general, `robotology-superbuild` is one of the many ways in which software is distributed, and hardcoding the use of a given generic software to a given distribution solution typically reduce re-usability. 

For an example, consider the existing README, it writes that: 
> It is recommended to install iDynTree, icub-models, OSQP and qpOASES using the robotology-superbuild:

it just suggests the use of robotology-superbuild, while this use of `ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX` basically makes this use compulsory. 

In general, the code in https://github.com/dic-iit/matlab-whole-body-simulator/blob/30642d6c195ae1c663d6ebbf43e81f95eff1946a/init.m is quite hardcoded to the use of iCub and robotology-superbuild. Could it make sense to change that and find the model via one of the many ways already supported by icub-models (see https://github.com/robotology/icub-models), for example by specifying the model via a `package://` URI?

We can think of having a code such as:
~~~
robotName = 'iCubGenova04';
[modelPath, fileName] = getFileFromROSPackageURI('package://iCub/robots/' robotName)
Config.modelPath = modelPath;
Config.fileName = fileName;
~~~

The `getFileFromROSPackageURI` function (that can be directly implemented in MATLAB) then can just find the file via the `ROS_PACKAGE_PATH` and `AMENT_PACKAGE_PATH` env variables (that are set if you install icub-models also via conda). See https://github.com/robotology/idyntree/blob/918084ff3ab4a9fb9e28c88f09dc6dbf9d6bddcc/src/model/src/SolidShapes.cpp#L200 for a related code.

nunoguedelha, 2021-03-26:
Yes, indeed, I saw the environment variables set in `conda/multisheller/icub-models_activate.msh`, among which `ROS_PACKAGE_PATH` and `AMENT_PACKAGE_PATH` as you said.

Initially I was planning to use them as you suggest, but avoiding to do a search of the file in those paths, I instead used directly the Yarp resource finder (via the MATLAB `system` command), which was also installed along with `icub-models` via Conda. This way, the model is searched the same way it is searched by the WBToolbox models.

I pushed the code in the branch `fix/for-using-superbuild-ninaries`. What do you think?

traversaro, 2021-03-30:
>  Fix any encountered issues. ONGOING: $ conda init failing on MATLAB online.

If this is tricky to solve we can also solve it later.

nunoguedelha, 2021-03-30:
> Yes, indeed, I saw the environment variables set in `conda/multisheller/icub-models_activate.msh`, among which `ROS_PACKAGE_PATH` and `AMENT_PACKAGE_PATH` as you said.
>
> Initially I was planning to [...]

Fixed by #20 .

nunoguedelha, 2021-03-30:
@traversaro , if it's ok for you we can close this

traversaro, 2021-03-30:
Ok, feel free to close yourself!

closed 2021-04-21 by nunoguedelha