ami-iit/matlab-whole-body-simulator #23: Fix issue with the meshes: convert from dae to stl

nunoguedelha, 2021-04-02:


traversaro, 2021-04-02:
> collada_urdf package has to be built and installed...

I think that that package converts a collada multibody model (that is different from an URDF model that use collada meshes) to a URDF model, to just convert a dae to an stl you can just use several available package online (see https://www.google.com/search?client=firefox-b-d&q=collada+to+stl) or software like meshlab.

nunoguedelha, 2021-04-06:
Actually, I was interested anyway in trying the `robostack` packages, and a catkin workspacefor building ros packages in case it would be requiired in the future.

Regarding the collada files, I assumed that the collada multibody model was an all-in-one file including also the meshes. In such case, if applied to a .dae file having only a mesh, the tool would eventually generate an stl mesh in the conversion process. A lot of ifs, I know, but I was just going to give a quick try.

traversaro, 2021-04-07:
> Actually, I was interested anyway in trying the `robostack` packages, and a catkin workspacefor building ros packages in case it would be requiired in the future.

Ack, however note that given that is not already built (see https://robostack.github.io/noetic.html) probably it has some kind of compilation failure.

nunoguedelha, 2021-04-07:
> to just convert a dae to an stl you can just use several available package online

I've tried https://anyconv.com/dae-to-stl-converter and https://3d-convert.com/en/convert/dae-to-stl.html. The first generates a binary file, while the later generates text file which can easily be previewed with Quicklook for instance (on MacOS). I kept the later.

traversaro, 2021-04-07:
> The first generates a binary file (which actually didn't work),

In the past to convert from binary to ascii STL I used https://github.com/cmpolis/convertSTL and it worked fine.

nunoguedelha, 2021-04-07:
Also tried meshlab, installing with brew: "binary damaged" 😒

commit ddacdaa0 (2021-04-08): Fixed #23: converted RRbot1 mesh from DAE to STL and added to model
PR #27 (2021-04-08): Add pendulum mesh (STL)
Fixes #23 .
Fixes #24 .
closed 2021-04-08 by nunoguedelha