ami-iit/matlab-whole-body-simulator #32: Add an example of using  matlab-whole-body-simulator  to run the YOGA++ controller from whole-body-controllers

traversaro, 2021-04-12:


traversaro, 2021-04-12:
https://github.com/dic-iit/matlab-whole-body-simulator/issues/32

nunoguedelha, 2021-04-26:
Task steps:
- [ ] Do a first test in `whole-body-controllers` replacing Gazebo by ["RobotDynWithContacts" + "IMU" + "robot visualizer"]
    - [ ] Create a "Robot Sensors" library and add a first "IMU" block.
    - [ ] Create a "Visualizers" library and add the existing "robot visualizer" block.
- [ ] Create a main model where the YOGA++ replaces the dummy one.

nunoguedelha, 2021-04-26:
@traversaro , I've updated https://github.com/dic-iit/matlab-whole-body-simulator/issues/32#issuecomment-826775664 according to what we agreed in the meeting. Moving now the issue (manually) to `whole-body-controllers`.

nunoguedelha, 2021-04-30:
Moved the issue to https://github.com/robotology/whole-body-controllers/issues/121. Closing this one.

closed 2021-04-30 by nunoguedelha