ami-iit/matlab-whole-body-simulator #52: Allow common `devel` and `master` commits history

nunoguedelha, 2021-05-28:
Hi @traversaro @Giulero @gabrielenava @CarlottaSartore .

So far, I have maintained two separate commits history, one for `devel` and one for `master`, and regularly merged `devel` to `master` or vice-versa. Usually, the `master` branch is for pushing fixes, and `devel` for implementing new features. This workflow is common in big projects, on repositories with many users and contributors. Sometimes we have both workflows, where `master` and `devel` can diverge temporarily but regularly converge to a single line of commits. In the case of this repository, the flow of issues and new features is quite slow for now and we could keep a workflow with a common commits history as often as possible. In such case, most merges between `devel` and `master` would be fast-forward when they don't diverge.

Do you guys have any opinion on this?

**Note:** There is also a new trend of renaming the `master` branch to `main` (https://about.gitlab.com/blog/2021/03/10/new-git-default-branch-name/). One of the repositories in robotology or dic-iit organizations adopted it, I don't remember which. I remember you being involved in this @traversaro , any news?

traversaro, 2021-05-28:
> Do you guys have any opinion on this?

We have several project that only have a `master` branch and no `devel` . The `devel` branch in general is much less useful now that we release much more systematically and people interested in stability can use the releases.  So, feel free to just use ``master` as a single branch, this specific choice is up to the maintainer.

gabrielenava, 2021-06-07:
> There is also a new trend of renaming the master branch to main (https://about.gitlab.com/blog/2021/03/10/new-git-default-branch-name/). One of the repositories in robotology or dic-iit organizations adopted it, I don't remember which

I've recently opened a new repo in the `dic` organization, and the default branch was `main` and not `master`. I don't know if you can choose between `main` and `master` or if the default branch is now `main`.

traversaro, 2021-06-07:
> I don't know if for new repos you can choose between `main` and `master` or if the default branch is now `main`.

I think the default is now `main` (also at GitHub level) but you can still switch back to use `master` (as you can use any branch name).

nunoguedelha, 2021-06-14:
I would then get rid of `devel`, keep only `master` and protect it (if it's not already the case).

But first let's cross-check with @CarlottaSartore nd @VenusPasandi if this doesn't disturb their workflow.

CarlottaSartore, 2021-06-14:
Thanks @nunoguedelha  for the ping! Since we are working with tags, I do not think this would have an impact on our environment, do you agree @VenusPasandi ?

VenusPasandi, 2021-06-14:
Thanks @nunoguedelha for your note! As @CarlottaSartore said we use a tag for this repo. So, it is fine for us.

nunoguedelha, 2021-06-14:
Great! I'll proceed then.

nunoguedelha, 2021-06-14:
Done.
- `devel` has been removed.
- `master` is now protected:
    - Branch deletion is not allowed to anyone.
    - Force-push is not allowed to anyone.
    - All PRs require at least one approving review before being merged.
    - A new review is required if a new commit is pushed.
    - All conversations resolution is required before merge.

**Note:** last commit in `devel` (2a553badd750e2d464d4def75c505da6efa01aa5) branch was a predecessor of current `master` branch commit, with exact same contents:

[image]

nunoguedelha, 2021-06-14:
This issue is ready to be closed.

closed 2021-06-15 by nunoguedelha