ami-iit/matlab-whole-body-simulator #53: Update the contact forces in the case of impact

VenusPasandi, 2021-07-13:
In the `robotDynWithContacts`, first the contact forces are computed using the maximum dissipation energy. Then, if an impact is happening, the velocity vector of the robot after the impact is computed. Then, the velocity vector after the impact and the contact forces before the impact are used for the forward dynamics!! But, in case of impact, the contact forces must be updated and the contact forces after the impact is used for the forward kinematic. The simplest point is that in the new contact point, just became in contact through the impact, the contact forces are considered zero!!

CarlottaSartore, 2021-07-16:
C.C. @nunoguedelha  @Giulero @traversaro

traversaro, 2021-07-16:
As discussed in f2f, probably @VenusPasandi can try to fix this problem and see if it creates problem. If it does not create problems, @VenusPasandi  can create a PR to fix this problem.