ami-iit/matlab-whole-body-simulator #54: Implement simulator block to handle closed loop dynamics

traversaro, 2021-07-16:
@VenusPasandi is working on an improvement for the simulation of mechanical systems with closed loop dynamics.

VenusPasandi, 2021-07-16:
yes, I am working on this in the branch [feature/constrainedDynamics](https://github.com/dic-iit/matlab-whole-body-simulator/tree/feature/constrainedDynamics). I will be happy to hear suggestions for improving it.

@nunoguedelha @Giulero

VenusPasandi, 2021-11-12:
I implemented the simulator block for the constrained dynamics which has two closed chain kinematics in d27c4a077706011f7b4523581b76a70bfbf62630. I need still to do some modifications:

- [ ] Provide the possibility of the having any number of the closed chains.
- [ ] Properly resolve the issue of dependent constraints for the closed chain kinematic,
- [ ] Properly resolve the issue of the inversion of the rank-deficient matrices,
- [ ] Clean the code

nunoguedelha, 2021-11-12:
Hi @VenusPasandi , I haven't take a look a this yet, I'll try to do it ASAP.

VenusPasandi, 2022-03-22:
I opened #64 to merge the simulator block for the closed-chain robots in to the master though we haven't addressed all the checkpoints in https://github.com/ami-iit/matlab-whole-body-simulator/issues/54#issuecomment-967248927. I think the remaining checkpoints are minor fixes that we can resolve them later.

closed 2022-03-28 by VenusPasandi

nunoguedelha, 2022-03-31:
Ciao @VenusPasandi , I saw you already opened one or a few issues tracking all the remaining points listed in https://github.com/ami-iit/matlab-whole-body-simulator/issues/54#issuecomment-967248927. I suggest you open an Epic to group those points, since the issue here is now closed.

VenusPasandi, 2022-03-31:
> Ciao @VenusPasandi , I saw you already opened one or a few issues tracking all the remaining points listed in [#54 (comment)](https://github.com/ami-iit/matlab-whole-body-simulator/issues/54#issuecomment-967248927). I suggest you open an Epic to group those points, since the issue here is now closed.

Ciao @nunoguedelha. Actually, the remaining points are not related to the closed chain dynamics but are general points. because of that I didn't group them. I am not sure if creating an Epic for all the improvements of the dynamics blocks is really usefull.

nunoguedelha, 2022-03-31:
I see, indeed there are only two remaining and directly related to the closed loop. Never mind.