ami-iit/matlab-whole-body-simulator #73: Do a new release

traversaro, 2022-04-14:
Hi @VenusPasandi, can I produce a new release of `matlab-whole-body-simulator` ? This would be convenient to fix https://github.com/robotology/whole-body-controllers/issues/144 . 
As if I understood correctly there have been breaking changes, I think we can release matlab-whole-body-simulator 3.0.0 .

traversaro, 2022-04-14:
cc @fabiodinatale

VenusPasandi, 2022-04-14:
Hi @traversaro ,

Yes, I think there is no reason for not producing a new release. Just small point, I forgot to check the compatibility of the last version of the `matlab-whole-body-simulator` with the `whole-body-controllers`.

traversaro, 2022-04-14:
> Just small point, I forgot to check the compatibility of the last version of the `matlab-whole-body-simulator` with the `whole-body-controllers`.

The current version is not working (see https://github.com/robotology/whole-body-controllers/issues/144) so there are two options:
* whole-body-controllers does not works with the new release (like now)
* whole-body-controllers does works with the new release

So, in the worst case we remain like now, best case it is fixed, so I think we can proceed.

VenusPasandi, 2022-04-14:
To be sure, I tested the whole-body-controllers and I found out some typos in the simulator. The typos are resolved in #75

closed 2022-04-14 by traversaro