ami-iit/matlab-whole-body-simulator #76: compute the contact wrenches in the sole frames

VenusPasandi, 2022-04-14:
In the simulator, we compute the pure forces applied to the contact vertices of the feet and then we convert them to the resultant wrenches applied to the feet soles and represented in the corresponding foot frame. In this regard, we use
\begin{equation}
\begin{aligned}
F = ^f R_w \right( F_1 + F_2 + ... \left), \\
T = Skew(^fp_1) ^f R_w F_1 + Skew(^fp_2) ^f R_w F_2 + ... \left), \\
\end{aligned}
\end{equation}

We use the above formula for the both rectangular and circular feet. But, it is not for the circular feet as the contact point in the circular feet is an instantaneous point and is not constant even in the foot frame.

DanielePucci, 2022-04-15:
@VenusPasandi the math seems not to render, cannot read

Please remember to add labels, history points, etc

VenusPasandi, 2022-04-16:
> @VenusPasandi the math seems not to render, cannot read

sorry, I corrected the math.

> Please remember to add labels, history points, etc

yes, sure