ami-iit/matlab-whole-body-simulator #77: Visualize an inclined surface in the robot visualizer

VenusPasandi, 2022-04-19:
Following #72, we want to provide the proper visualization of the motion on an inclined surface.

As I discussed with @traversaro, the idyntree visualizer can not visualise an inclined surface. But, @traversaro suggested to add a joint-less model representing the inclined surface.

cc: @Giulero @valentino-razza

fabiodinatale, 2022-06-13:
In order to achieve the task, which will be carried out by me with the support of @VenusPasandi, I would ask about the following possible solution which part of the code should be modified.

> As I discussed with @traversaro, the idyntree visualizer can not visualise an inclined surface. But, @traversaro suggested to add a joint-less model representing the inclined surface.

Could it be possible to introduce this plane as an object of the visualizer rather than introducing it as a joint-less model?

CC @traversaro

traversaro, 2022-06-13:
> In order to achieve the task, which will be carried out by me with the support of @VenusPasandi, I would ask about the following possible solution which part of the code should be modified.
>
>
> Could [...]

If I recall correctly the visualizer, there is no concept of "object" that is differeny by joint-less model.

closed 2022-07-25 by fabiodinatale